MarkJelic/TEC-1G #16: Speaker enable jumper

JamesElphick, 2023-09-17:
We could add a simple two pin jumper nearby to physically disable speaker connection.

Ideally not directly underneath LCD module when installed so that it won't require removal.

MarkJelic, 2023-09-24:
Added a "Groundwalker" jumper that will be initially hard traced together. (So no jumper actually needed and will not be supplied in the kit.)

But if you want to yell "SILENCE, Groundwalker!" at your TEC-1G, you can cut the trace between the jumper holes, solder in a pair of pins and then install or remove a shunt at your will.
[image]

closed 2023-09-24 by MarkJelic

MarkJelic, 2023-09-29:
Actually... This works better..
[image]

MarkJelic, 2023-09-29:
And this is the PCB.  Issue now closed.
[image]