MarkJelic/TEC-1G #17: TEC-1G Schematic v1.7 readability.

tinkerzone, 2023-09-19:
While looking very cool, the TEC-1G v1.7 Schematic isn't readable on a Single A4 Page. While potentially readable on an A3 page, and on a large monitor, I think we need to keep with the TEC's background as stemming from a magazine article and so it needs to be printable on A4. Suggest splitting over 2, 3 or even 4 pages.

MarkJelic, 2023-09-19:
Ooof... That a bit of a job... But yeah, I get why you suggest that. (And yes, It is in fact done on one large A3 page.)

So... Do you have any suggestions of how it should be broken up?

tinkerzone, 2023-09-20:
This could be a schema:
Cpu/clock.
Keyboard and hex pad.
Displays and i/o.
Memory and Memory management.
Power and expansion.