MarkJelic/TEC-1G #19: Additional Silkscreen Issues

tinkerzone, 2023-09-19:
MEMBUS - Header
IOBUS - Header
GPI/O  - Header
CONFIG - Switch
STATUS - Bar
JP5 - Header
Z80Bus Female - Socket
Z80Bus RA - Socket

The silkscreen text related to each of these 8 parts are too small and not consistant with the other text on the board. 

MEMBUS - Shift Room Size Jumpers, u14, and Chip Size Jumpers down PCB. Increase text size.
IOBUS - Shift header 4mm to the left. Increase size.
STATUS - Increase size with no drawbacks.
GPI/O - Shift GPI/O label down 3mm. Increase text.
CONFIG - Increase text. Break the LCD silkscreen.
JP5 - Bruh?! - Increase text.
Z80Bus Female - Increase text of RD/IORQ/A10/A11.
Z80Bus RA - Increase text of RD/IORQ/A10/A11.

[image]
[image]
[image]

[image]

MarkJelic, 2023-09-19:
This is not going to be an easy fix... Sure, the labels might be able to be made a smidge bigger, but I really doubt they can get to the same size as the rest of the labels. (And there is a consistency... they are all the same small size :)  And if I can slightly increase the size of some but not others, then there really will be inconsistency.

Another option could be to put the labels on the underside?

tinkerzone, 2023-09-20:
So Config switch and status bar labels need to be viewable by the user. Would suggest increasing the size of these labels is non-negotiable.

I'd be fine if the labels for the headers were kept small on the top but on the bottom side larger. 

There is a large amount of space on the bottom side where additional information can be silkscreen. I have seen other boards that have like logic tables that show what memory is selected etc when what jumpers are set. See the N8VEM sbc project board.

MarkJelic, 2023-09-30:
1. None of the headers can be moved. We have cards relying on their position to connect correctly.

Have increased the size of the Config and Status labels slightly. Any bigger and they would either conflict or start to be hidden by components.

[image]
[image]

Have added some labels to the back:
[image]

closed 2023-09-30 by MarkJelic