MarkJelic/TEC-1G #20: Schematic V1.8 - Switch 11/12 Clarity Issue

tinkerzone, 2023-09-19:
Rename Switch 11 and Switch 12 Switch 11a and Switch 11b and/or put a dashed line in-between to indicate they are mechanically connected.

MarkJelic, 2023-09-24:
Changed the switch to something custom, as previous would not have worked anyway.

[image]

tinkerzone, 2023-09-24:
Hi Mark,
Can I suggest this description goes on the silk screen overlay on the bottom side of the board as well? The N8VEM project puts info about jumpers on their boards and it's very handy.

tinkerzone, 2023-09-24:
[image]
Mark, this is a more elegant solution to this issue.

MarkJelic, 2023-09-24:
Although I agree that is very elegant (and very space efficient) I think it is too difficult, without exactly those pictograms printed somewhere, for people to easily switch from one to the other. 

As for the text on the underside... I think that should be doable.

Is there any other text that needs printing on the underside?

MarkJelic, 2023-09-24:
I'll close this off as somehow its turned into a duplication of Issue #22

closed 2023-09-24 by MarkJelic