MarkJelic/TEC-1G #22: The decoding for the EXPAND socket needs confirmation

MarkJelic, 2023-09-23:
Beta 1 did not support 24 pin chips in the EXPANSION socket.

[image]

So for Beta 2, the jumper setup used for the ROM socket was copied to the EXPANSION socket, with the only difference being that the switch is controlled by the EXPAND system latch bit.

[image]

But with testing of Beta 1, it was identified that 27c256 EPROMs only allow for the top half of the rom to be allowed to be seen, due to pinout difference of the 27c256 to the 28c256 EEPROM which is what many of us were testing with. Thus it was unlikely that 27c256 worked in the EXPANSION socket properly, either.  Meanwhile, it was discovered that 27c512 (a 64k UV EPROM with only 28pins) could be successfully used in the ROM socket, if it's A15 pin was also allowed to be switched. So an extra switch was added to the v1.8 of the schematic (proposed for the Production run) as below. Conveniently, this ALSO fixes the pinout difference of the 27c256 (we hope) to thus allow you to manually select which half of the 32k EPROM you want to see.

[image]

So for the production version, it would be good if the EXPANSION socket also was compatible with 32k EPROMS, be they either the UV sort (27c256) or the EEPROM sort (28c256) and be able to be controlled by the EXPAND bit. But the pinout differences would require the SWAPPING of a couple of pins between those toe EPROM sorts. Hence, Mark came up with this:

But after asking for confirmation, Craig suggested this, but he had in mind support for the 27c512, which is not the intention of this socket. Only a maximum of 32k is to be paged in this socket, and to be frank, it is largely intended for RAM to be installed, not ROMs. The only reason to allow ROMs in this socket is to allow "copying" of any ROM (up to 32k) you may find in the wild, or designing a program that needs 32k of space for its code.  Here is Craig's suggestion... But this, as with Mark's above, needs to be thoroughly thought through, before we go to production, without testing!

[image]

1971Merlin, 2023-09-23:
Mark, your final design allows for /WR to be jumpered to pin 1. This is incorrect. No device uses pin 1 for WR. 

Pin 1 is either A14, no connection, or +5v depending on chip type. Your schematic needs a re-work.

MarkJelic, 2023-09-23:
> No device uses pin 1 for WR.
the 27c256 uses VPP on Pin 1... That's kinda like WR... But obviously not enough to actually program it so I don't think it would do any harm to have WR connected to it.

I've updated the graphic to show it is a DPDT switch that works in unison, basically swapping the connections on Pin 1 and Pin 27 from all other 28 pins chips, to the damn 27c256.

It still looks to me as if my circuit should do the trick.

IanMcLean68, 2023-09-24:
> the 27c256 uses VPP on Pin 1... That's kinda like WR... But obviously not enough to actually program it so I don't think it would do any harm to have WR connected to it.

Probably not. The minimum programming voltage you seem to find on windowed eproms is 12V. It is more conventional though to have Vpp going to ground when not being programmed.

MarkJelic, 2023-09-24:
OK, how about this?  When the DPDT switch is on 2-3/5-6, it is in 27C256 mode. Up for all other chips.
[image]

MarkJelic, 2023-09-24:
I made a custom footprint for it :) 
[image]

MarkJelic, 2023-09-24:
In circuit
[image]

tinkerzone, 2023-09-24:
This is my solution to sw5.

[image]

tinkerzone, 2023-09-24:
Also, as per my comment on issue 20 would be helpful to have the jumper option silkscreened on the bottom of the pcb.

tinkerzone, 2023-09-24:
Actually, this version of sw5 would allow a dpdt switch to be used.
[image]

tinkerzone, 2023-09-24:
[image]

MarkJelic, 2023-09-24:
Ooh... I like this solution... Why could nobody else come up with it??  OK, let me try that in the schematic and make sure it makes sense...

(You know I was wracking my brain, over and over, saying to myself "I am SURE there is a way to work with with a dpdt switch, but I just couldn't make it come together!)

tinkerzone, 2023-09-24:
:D

MarkJelic, 2023-09-24:
Yep, I think it works! Well done!! :) 
[image]

tinkerzone, 2023-09-24:
Woohoo!

MarkJelic, 2023-09-24:
OK, and the final setup, that also allows for the use of 6116 (and equivalent NVRAM packages) is this:
[image]

Wont close this until the PCB is routed.

1971Merlin, 2023-09-25:
I've had a look over the "OK, and the final setup, that also allows for the use of 6116 (and equivalent NVRAM packages) is this:" version, and this seems to work correctly for all chip types.

MarkJelic, 2023-09-25:
>Vpp is usually held at +5v when not programming, however. The following is from the 27c256 datasheet. To meet this, connect SW5A leg 3 to 5V, not GND.

No harm and easy to fix. I'll update the schematic.

MarkJelic, 2023-09-26:
[image]

Fixed... Now doing the PCB

MarkJelic, 2023-09-30:
So this is the final setup of the Schematic:
[image]

And this probably wont help, but this is the bit of the PCB as routed:
[image]

And this is the Render:
[image]

closed 2023-09-30 by MarkJelic