MarkJelic/TEC-1G #25: DIP-8 analogue switch and fast clock redirect

JamesElphick, 2023-09-29:
### **FAST CLOCK REDIRECT**

Beta 2 design for the clock switching circuit relies on the analogue chip (currently TS5A3159DCKR "The Tick") being operational in order to transmit the fast clock. This is because the fast clock is connected to the NO (Normally Open) input of the switch.

Current design:

[image]

The fast clock will be required for most use cases. So it would be ideal to make the fast clock less reliant on the switch by instead routing to the NC (Normally Closed) input.

Changes to redirect fast clock include:
- Redirecting the Fast clock to NC pin (currently Pin 3) of analogue switch.
- Redirecting the Slow clock to NO pin (currently Pin 1) of analogue switch
- Moving GND to High/Low pin (currently Pin 6) of analogue switch
- Moving +5V and R32 10K resistor to Pin 2 of GaGa switch

With GND normally applied (GaGa switch Open) this will enable the Fast clock by default.

Proposed design (changes highlighted):

[image]

### **DIP-8 ANALOGUE SWITCH**

The current "Tick" analogue switch has been found to have capacitive coupling or crosstalk issues. A DIP-8 alternative is being considered to allow use of an alternative analogue switch (in DIP-8 package) and other package options for experimentation using an add-on PCB.

Best option right now is the MAX4544 chip with this pinout:

[image]

ATTiny85 microcontroller also comes in DIP-8 package as an option with +5V and GND on the same pins.

[image]

However Reset on Pin 1 of the ATTiny85 means that these two chips cannot use the same socket. Reset line would also need to be routed to the chip for this purpose. This option of a microcontroller could be revisited with a future board revision.

Changes for DIP-8 analogue switch include:
- Routing +5V to Pin 8 on MAX4544
- Routing GND to Pin 4 on MAX4544
- Routing Pin 1 of GaGa switch to Pin 7 on MAX4544
- Routing the CLK line to Pin 2 on the MAX4544
- Pin 5 and Pin 6 of MAX4544 will be N.C. (No Connect)

Proposed design for MAX4544 (changes highlighted):

[image]

[image]

MarkJelic, 2023-09-29:
OK, so here is the updated schematic:
[image]

And I've done something a bit radical... Since you either use the manual switch or you use the Tick, and not both at the same time, I've overlayed one over the other.

[image]

Thoughts?

JamesElphick, 2023-09-29:
That's an awesome idea with overlaying both footprints!

One change I would make is to move the R32 10K resistor to the +5V side of the GaGa switch. That way Pin 7 of the 4544 is normally GND, and doesn't receive the full power rail current when the switch is on, just a +5V logic signal.

[image]

MarkJelic, 2023-09-29:
Mmm... That contravenes my understanding of how pull up/down resistors work. In my (amateaur) understanding, that line would never go high. The direct ground link would always pull it low, no matter what the switch is doing.

I'll wait for others input on this.

MarkJelic, 2023-09-30:
PCB routing completed.  Testing will have to be on the production run.  Issue closed.

closed 2023-09-30 by MarkJelic