MarkJelic/TEC-1G #36: switch SK-22F39 or SK-22F24 impossible to find

mciucci65, 2024-09-03:
Hi great project but I can't find on this type of switch. Do you have some equivalent or other site where I can find ?.

On Aliexpress there are not.

kind regards

MarkJelic, 2024-09-05:
Yes, the power switch we selected initially is now seemingly impossible to source. So I have made a little adapter board which is available for US $1 plus postage. (Best you buy another kit and then save on postage.) Here is a link to the instructions for installing it.
https://github.com/MarkJelic/TEC-1G/blob/main/files/Installing_New_Power_Switch.pdf

The item is still being approved on Tindie, but here is the link to it when it does:
https://www.tindie.com/products/tec1/tec-1g-replacement-power-switch/

mciucci65, 2024-09-05:
thanks for quick replay when will be available the gerber files for this and RTC,SD graphic display module ?

MarkJelic, 2025-03-19:
I'll get the gerber up for this and all the others by the end of the week.

mciucci65, 2025-03-24:
hi Mark,

thank you forcyour great job
do you have ant  news about gerbers upload ?

kind regards

mciucci65, 2025-03-24:
Hi Mark,
Thanks for RTC gerber upload can you also upload SD gerber

kind regards

MarkJelic, 2025-03-24:
RTC and SD Gerbers are up.

The new power switch adapter PCB is here:

https://github.com/MarkJelic/TEC-1G/blob/main/files/TEC-1G_Power_Switch_v11.zip

closed 2025-03-24 by MarkJelic

mciucci65, 2025-03-24:
Thank you very much for your great job and for gerbers upload.

kind regards
MC

mciucci65, 2025-04-05:
Hi Mark,

Thanks again for the gerber uploaded.

I need more information about MON 3 1.6. I see in the Pata document on
github that you mention it but i cannot find it.

Where I can download it.

kind regards
MC

MarkJelic, 2025-04-05:
This has not yet been officially released but I believe you can get it from Brian Chiha who is working on it.

https://github.com/bchiha/Ready-Z80/tree/main/26-PATA_FAT32

mciucci65, 2025-04-05:
thanks !!!
I checked the sd gerber  and you forgot to upload the 2 mini pcb output
display and input selector.

Can you upload also these pcb ?

kind regards

bchiha, 2025-04-05:
Hi,  

Mon3 v1.6 hasn't been released yet, I still have a few things to add.  The PATA code to read a disk can be found in the link Mark provided.  But, if you contact me, I can provide you with the current Mon3 1.6 with the PATA code.  However, it's only useful if you have the PATA PCB installed.

I do plan to add the functionality for disk reads for the Micro SD Add-on too.

mciucci65, 2025-04-05:
hi,
thanks for quick response i will wait for sd support in new mon 1.6.

thank again for your great project.

kind regards
MC

MarkJelic, 2025-04-05:
I have uploaded the Input Tester Gerber. I never had a separate Output tester on it's own. I originally had it as part of the SD car to test PCB printing with separate snap-off PCBs. But the PCB printers charge some stupid extra amount for this option so I quickly dropped it for separate little boards like the Input tester. However, but then I had also found on Tindie a very capable and cheap output tester board that was cheaper than I could put it all together and so that is what I have been supplying for a while now. This is what I found

https://www.aliexpress.com/item/1005004567063273.html

Now... About this adding comments to already closed Issues and even Issues that are totally unrelated.
For example, what has MON3 v1.6 got ANYTHING to do with the old power switch not being able to be found??
It is not fair anyone having to find which subject you randomly decided to add a comment to.

Please, no more. Stick to the topic of the Issue, at a very minimum. And ideally, if it is a NEW Issue, then please create it as a NEW Issue.  I don't close Issues off until I think they are in fact completely dealt with. In the case of these testers, yes would have been appropriate to tack it onto the end of the issue you raised about the Gerbers for the RTC and SD... Not in this one.  And certainly not throwing in MON3 1.6 questions. That is inappropriate because NOBODY else except you and Brian/Me would find this question!  Please start a NEW Issue any questions regarding MON 1.6. Any future breaks in this common sense approach will be deleted and ignored.