MarkJelic/TEC-1G #37: Gerber Files for RTC and SD

mciucci65, 2024-09-16:
When will be available these gerber files for RTC e SD module ?

MarkJelic, 2025-03-19:
I just need to update them with the new 2x10pin header and then I will put them up on here. Give me till the weekend.

(Sorry, I don't get any notifications about Issues being raised, so I didn't see this until now. I'll make sure to keep an eye on this from now on.)

mciucci65, 2025-03-19:
Hi Mark great news,

thank for update and wait weekend for gerber upload.

kind regards

MarkJelic, 2025-03-24:
Done.

closed 2025-03-24 by MarkJelic

MarkJelic, 2025-04-05:
I have uploaded the Input Tester Gerber. I never had a separate Output tester on it's own. I originally had it as part of the SD car to test PCB printing with separate snap-off PCBs. But the PCB printers charge some stupid extra amount for this option so I quickly dropped it for separate little boards like the Input tester. However, but then I had also found on Tindie a very capable and cheap output tester board that was cheaper than I could put it all together and so that is what I have been supplying for a while now. This is what I found

https://www.aliexpress.com/item/1005004567063273.html